nordicsemi/IOS-nRF-Connect #1: writing/sending a value for a characteristic

hoLiFuk, 2016-08-25:
Hi. I'm new to the BLE world, so please pardon my naivete.

I'm trying to write a value for a BLE Characteristic using your nRF-Connect ios app. The BLE service is this Arduino 101 example: https://www.arduino.cc/en/Tutorial/Genuino101CurieBLECallbackLED

The screen for writing a value is a bit different on the ios app, relative to the Arduino link above -- it's probably a newer screen. The current screen shows "Write value" and "Write types." It's letting me type a value in "Write value" but nothing is happening when tapping the area for "Write types."   Obviously, tapping "Send" isn't working. 

Am I doing something wrong?

(In the attached image, the value typed is "0" but it doesn't really matter. 1 or 01 isn't working either.)

[image]

mostafaberg, 2016-08-26:
Hi!, you are correct they look different, the screenshots on the arduino page are actually showing the Android app, not the iOS, we currently have minor differences that we are planning to resolve in the near future!

regarding your issue, are you sure you're still connected to the peripheral before tapping write value ? also, is there any specific entries in the logs before you tap send and after you try to tap send ?

mostafaberg, 2016-08-26:
And about the write types from what I recall is the app only supports Request type, we will either hide this completely or implement the feature correctly as you see Command is disabled, which might be confusing

hoLiFuk, 2016-08-27:
Thanks for your reply! Tried the Android app (everything else the same), and it works -- writes 1 or 0, the LED turns on or off, etc.

On the iPhone, there is no change in the log after a value is entered (0 or 1) -- the "send" is not accepted, the text box becomes red. I haven't been able to type anything for "Write types."

Since the Android version is working, this is not an issue for me anymore, but if you want me to test something else, happy to do so.

Here is the ios log:

[image]

mostafaberg, 2016-08-28:
Happy to hear your resolved the issue, we are currently aware of some issues in the iOS app and will be undergoing some fixes soon, so will surely message you if we need some testing or help!

closed 2016-09-05 by mostafaberg

dakins1, 2019-10-12:
Hi, I am using an iPhone 6s with the latest version of nRF Connect. I am having the same issue as described above. Both request and command do not work from my iPhone. However, it works perfectly fine with an android. How can I get this to work with my iPhone? I am using an nRF52810