nordicsemi/IOS-nRF-Connect #3: iOS app crashes on app startup.

klyhthwy, 2017-06-06:
I am seeing this issue on an iPhone 7, an iPad mini 2 with Retina Display (model ME280LL/A), and also an iPhone 6 (or 6s). The iOS version of the iPhone 7 and iPad are 10.3.2. The nRF-Connect app is the latest in the App store. The apps open and crash immediately making it unusable. We have tried force closing it and even deleting the app and reinstalling with no improvement.

With past apps you have released I seem to remember there may have been an issue with an overwhelming amount of radio congestion causing issues. Here in Logan, Utah at Icon Health & Fitness we do have this kind of environment in our development office.

klyhthwy, 2017-06-06:
I have noticed that if I turn off Bluetooth on my iPhone 7, I am able to open the app and start a scan for a second. But when trying to connect to a device or even view its advertising data the app crashes.

jovirus, 2017-06-07:
Hi @klyhthwy, I suppose this issue is caused by some incorrect advertisement data, while nRF Connect was trying to parse them. I have fix an similar issue. A update will be included for next release..

closed 2017-06-09 by jovirus