nordicsemi/IOS-nRF-Connect #4: log verbose yellow is illegible

DG12, 2017-06-16:
Please revise the log to use a different color for verbose than yellow(gold?)

It would be helpful to have the second and subsequent lines for each event to start under the time (not indented ) allowing more viewable data on the screen,

A nice  idea might be an option to show time in  milliseconds since connect or milliseconds since last entry.

Thank you

jovirus, 2017-06-19:
Hi @DG12 Thank you for suggestions, what is the motivation of using other colours than yellow? 
I think the alignment of the log text are looking good, since it give a intuitive way of displaying time sequences and respective message. 
The format of time [HH:MM:SS:**mmm**] 
The last three digits is in millisecond

DG12, 2017-06-23:
re yellow: contrast is very bad and cannot be read.

jovirus, 2017-07-18:
Thank you for suggestions

closed 2017-07-18 by jovirus