nordicsemi/IOS-nRF-Connect #5: pull down to clear is not the best

DG12, 2017-06-16:
When there are several beacons on the screen and you are scrolling down to see more, 
Scrolling up to see the previously listed ones may clear the screen 
Perhaps a "clear" button might be placed  next to "chart"

Thank you

jovirus, 2017-06-19:
@DG12 I have notice the problem, in future version the listed devices will be given in a sorted list, (A-Z) for example, to make it easier to find wanted device. 
You can drag the tableview down to refresh the list, in the current version 1.7.5

DG12, 2017-06-23:
Dragging the tableview to refresh IS the problem. I see 5 beacons on the list. I know there are more. I scroll to see the additional ones. I then realize the one I am looking for must be on the prior set. I scroll to see previous ones and the list is cleared. 
Sorting the list will not help this issue

jovirus, 2017-07-18:
We will work on to improve this.

closed 2017-07-18 by jovirus