nordicsemi/IOS-nRF-Connect #8: How does nRF Connect uniquely identify an iBeacon through CoreBluetooth?

tomdaniel0x01, 2017-08-23:
I am creating an application that needs to connect to an iBeacon, however CoreBluetooth does not provide `kCBAdvDataManufacturerData` for iBeacons or any other identifiable information that can be used to correlate a discovered BLE device with a discovered iBeacon device. Also the identifier that CoreBluetooth provides changes every time a new scan is started.

I noticed that your application is not only able to connect to an iBeacon device, but also retain persistent information about the device even after the app has been deleted and reinstalled. Could you please provide some insight into how these two things are possible?

This would be a life saver and greatly appreciated!

Thanks,
Tom

philips77, 2017-08-24:
Hi,
nRF Connect does not have access to iBeacons' manufacturer data yet (afaik). Perhaps you have Nordic "nRF Beacons"? They are very alike, but have Nordic's Company Identifier and then CoreBluetooth reads all data normally. iBeacons are handled separately by CoreLocation and it strips the beacon part out from the adv packet. We were planning to add a "beacons" screen one day, but that's a distant future. If I'm mistaken, @jovirus can have a better answer.

jovirus, 2017-08-24:
@tomdaniel0x01 I want to correct your saying, 1. you can not connect to a beacon; 2.iBeacon APIs are listed under coreLocation; 3. "kCBAdvDataManufacturerData" is not using for parsing iBeacon manufacture advertising package, please follow the Apple iBeacon standard you find it [here](https://developer.apple.com/ibeacon/);

What do you mean by retain persistent information, which?

tomdaniel0x01, 2017-09-01:
Thanks @philips77 and @jovirus for the fast response!

I am happy to say that have resolved my problem. I am able to use the `name` field of BLE to uniquely identify my BLE/iBeacon device with CoreBluetooth and associated the CoreBluetooth UUID with a CoreLocation beacon (We have programmed our own iBeacon device to make it connectable).

The reason I reached out here is because I thought nRF Connect was able to identify our device as an iBeacon, but it looks like I was mistaken. I now see that it is only listing BLE devices.

jovirus, 2017-09-04:
@tomdaniel0x01 nice to hear you have solved your problem.For ibeacon, please use [nRF Beacon](https://www.nordicsemi.com/eng/Products/Nordic-mobile-Apps/nRF-Beacon-App). I might be we adding iBeacon feature into nRF Connect as well in the future.

closed 2017-09-04 by jovirus