nordicsemi/IOS-nRF-Connect #11: Does not parse iBeacon or Eddystone formats

KenOgami, 2017-10-16:
Previous versions would display a dialog box to select Eddystone formats and display the URL for URL format packets. Version 1.8 does not. Version 1.8 alao does not parse iBeacon packets at all, all fields show N/A.

Version 1.8 only shows the default advertising fields and does not parse the advertising data as the Android version does so nicely.

jovirus, 2017-10-23:
Hi, @KenOgami Thank you for reporting this. If you have a failing case to parse your beacon data, please attach it here.

jovirus, 2017-11-02:
Problem solved in release v1.8.1

KenOgami, 2017-11-02:
In testing v1.8.1, when a beacon is advertising Eddystone format, nRF Connect immediately crashes.

jovirus, 2017-11-03:
@KenOgami I am doing an update now, if you want to test it, please send your AppleID to us mag@nordicsemi dot no

closed 2017-11-09 by jovirus