nordicsemi/IOS-nRF-Connect #15: Disconnected on iPhone 8 when connected

Xanthuim, 2017-11-07:
I use the iPhone 8 to connect the Bluetooth chip, soon disconnected. The output error message is as follows:

  [System] Error update value for charact. FA01 Reading is not permitted.
14: 06: 46.401 [System] Error update value for descriptor 2902 Reading is not permitted.
14: 06: 46.461 [System] Error update value for descriptor 2902 Reading is not permitted.
14: 06: 48.239 [System] Error disconnect from peripheral SHOWBABY_05: 00: 80 The connection has timed out unexpectedly.
There is a UUID Bluetooth device is FA01, no read permission.

jovirus, 2017-11-07:
Hi, @Xanthuim 
From the report you are given, I found there are at least two spots need to be inspected. First, descriptor 2902(Client Characteristic Configuration) shall provide read and write permission; [read here](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml)
Second, you are trying to disconnect from your peripheral, but it times out unexpected. It made iOS force to disconnect. 

I would suggest you check on your firmware.

mostafaberg, 2018-01-24:
@Xanthuim You can't write on 2902 on iOS, but you can enable/disable notifications which does the exact same thing, so if you want to set `0x01` to descriptor `2902`, then enable notifications for it's characteristic, iOS will decide to set the notification or indication flag, but will always choose the notification flag first.

if you elaborate a bit more what you are trying to do and what those services are we can help you more. in your case, the peripheral has disconnected from you after a while, could be connection interval related, might be pairing related, we can't know.

mostafaberg, 2018-07-09:
Is the issue still reproducing after the latest update?
if there's no update on this issue during this week I'll be closing the issue as resolved :)

Xanthuim, 2018-07-10:
e, sorry……, I have not paid attention to this problem.

closed 2018-07-10 by Xanthuim

mostafaberg, 2018-07-10:
no problems at all :) glad it has been resolved !, please open up new issues if you experience any more troubles!:)