nordicsemi/IOS-nRF-Connect #16: Make DFU rename optional

ambrice, 2017-11-18:
Recent nRF Connect update made it so I can not do a DFU update to my device because of bugs in the SDK's rename functionality.  Can you make this an option?  Details:

https://devzone.nordicsemi.com/question/179272/nrf-connect-ios-dfu-fails-for-unbonded-buttonless-dfu-service/

philips77, 2017-11-18:
Hi, I answered in a comment on the link you provided. It's a matter of DFU library rather then nRF Connect, as this functionality is built in into the library. We do not plan to have the renaming feature optional, as it ensures that the iPhone will connect to the right device during the DFU process. Please, fix the issue on your side, or return Invalid opcode when 0x02 request is sent, as it used to be in SDK 12 and 13.

ambrice, 2017-11-20:
I can fix the issue on my side, but I can't update the device with the fix.  So to summarize:

1. We wrote our buttonless DFU application per the (incorrect) documentation
2. We tested it on Android and iPhone, working fine.
3. We shipped it to the customer, did several updates successfully using nRF Connect
4. After latest nRF Connect update, customer can no longer use it to update device, due to a bug in the SDK where an async SVC call is made without checking if it's initialized first, causing a hard fault.
5. I can update the app with a fix, and fix the SDK, but I have to tell the customer to go buy an Android phone to be able to update to it.

philips77, 2017-11-20:
Ok, I get it. Do you use nRF Connect to update the devices, or have your own app, where you could use the updated DFU Library with option to disable this feature?

ambrice, 2017-11-20:
We are using nRF Connect.

jovirus, 2017-11-22:
@ambrice 
We can make a setting menu for DFU, so you can choose to disable the renaming function.
I hope that will work.

jovirus, 2017-11-29:
@ambrice I would like to invite you for testing the new feature when using DFU, which gives you option to switch off the renaming function. If you have appleid, I add you into our Testflight program.

ambrice, 2017-11-30:
@jovirus It's apple.developer@datasoft.com.  Thanks!

jovirus, 2017-12-01:
@ambrice You are now invited on Testflight.

ambrice, 2017-12-01:
Tested it and working well, thanks.

closed 2017-12-05 by jovirus