nordicsemi/IOS-nRF-Connect #17: Not find nrf_ble_gzl multiprotocol on iPad iOS 9.3.5

bootchk, 2017-11-29:
Running the examples/multiprocol/ble_gzl on an NRF52DK (built using SES on Linux), NRF Connect on an iPad does not find the advertisement, whereas on my iPhone 4, NRF Connect does find the advertisement.   I turned Bluetooth on after starting NRF Connect in both cases.   Not a big deal, maybe there is some other setting on the iPad, or I have exhausted some resource.  NRF Connect shows "Suspended" in gray in the upper right, and that never changes to "Scan" like it does on the iPhone.

jovirus, 2017-11-29:
@bootchk Thanks for your feedback
Which iPad model are you using? 
Apple starts to support BLE from iPhone 4s and above, iPad 3rd generation and above.

bootchk, 2017-11-30:
Oops.  Yes, my iPad is gen 2 and iPhone is 4S, so that explains it.

closed 2017-11-30 by bootchk