nordicsemi/IOS-nRF-Connect #18: advertise data change after app enter background/iphone screen off

macrogiga, 2017-12-19:
as Advertiser, enable app to send normal adv data. press power button to let the iphone screen off(or app run in background), then the adv data will change to "02 01 1A 14 FF 4C 00 01 0A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00" which is not relevant to normal adv data.
How can the app keep sending the normal adv data after entering background?

macrogiga, 2018-01-23:
Thanks Jovirus. Advertising the same packet after turning in background(even for a short time) is user-friendly. Do you have any plan to implement this feature in ios nRF connect app? Thanks.

mostafaberg, 2018-01-24:
@macrogiga This is an expected behaviour and it's out of our hands to modify how iOS handles that, the easiest way is to keep the phone from sleeping :) if you have any further questions please open up a new ticket and we'll gladly help!

closed 2018-01-24 by mostafaberg