nordicsemi/IOS-nRF-Connect #19: source code of nrf-connect

Demonsu, 2018-01-09:
Hi,

nrf-connect app could find ble devices which I can't see in my own apps. And after connecting those devices in nrf-connect app, they will show up in my own app.

I was wondering whether I did anything wrong in my code. Could you please share your source code or could you enlighten me what's wrong during my developing of ble.

Thanks,
Fan

philips77, 2018-01-09:
Seems like you are obtaining connected devices from the system instead of scanning for advertisements. The source code for nRF Connect is not public, however, we do release code of nRF Toolbox app (https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/) and nRF Blinky app (https://github.com/NordicSemiconductor/IOS-nRF-Blinky) where you'll find code responsible for scanning.

closed 2018-01-09 by philips77

Demonsu, 2018-01-15:
Thanks Aleksander! That really helps.
But another question is that retrieveConnectedPeripheralsWithServices requires non-null serviceUUIDs as parameter. This requirement limited the search within our own peripherals, but in your nrf-connect app, it could find all peripherals. How did you do that, is there any default UUID which could cover any peripheral? Please enlighten me, thanks!

philips77, 2018-01-15:
The method you mentioned is to retrieve already connected devices. If you want to scan for advertising ones, use CBCentralManager.scanForPeripherals(...). There you may give nil as service to obtain all advertising devices.