nordicsemi/IOS-nRF-Connect #20: Delete DFU Packages

GoodnatureBeacon, 2018-01-23:
I'm using the DFU functionality of nRF Connect and it would be really good to be able to delete old packages.

jovirus, 2018-01-23:
Hi @GoodnatureBeacon You can swipe left to delete firmwares.

GoodnatureBeacon, 2018-01-23:
I meant under the "Distribution Packet (Zip)" Menu not the "My Firmwares" menu. Packages get added to this when you send them to nRF Connect, but there is no way of deleting them

mostafaberg, 2018-01-24:
@GoodnatureBeacon Sorry about this mistake, I can confirm it's a bug on our side and will be fixed in the next release, problem is there's an invisible view that's over the approximate location of the third raw, causing it to block touches. to overcome that, you may try delete the firmware abote it first to move it away from the invisible view's location.

GoodnatureBeacon, 2018-01-24:
Just for reference, this is a seperate issue to the Third Item issue. This issue refers to not being able to delete Firmwares under the "Distribution Packet (Zip)" menu, not the "My Firmwares" menu where this was an issue.

mostafaberg, 2018-01-25:
@GoodnatureBeacon of course!, i missed that point, will reopen the other issue again

jovirus, 2018-05-30:
fix in V1.8.5

closed 2018-05-30 by jovirus