nordicsemi/IOS-nRF-Connect #25: Secure buttonless DFU is Abort

avhtak, 2018-03-29:
Hi

I tested the buttonless secure DFU with this tool, but an error(display “DFU service not found” and Abort) occurred.
This problem does not occur on Android phone and the nRFToolBox for iOS.

iOS(11.2)    nRF Connect v1.8.2… NG(Display “DFU service not found” and abort )　
                  nRF Toolbox v4.4.4… OK

Android(7.0) nRF Connect v4.17.0… OK
                     nRF Toolbox v2.6.0… OK

I tried again after turning off BT and iPhone, but it did not improve. 
Could you please advise me about this problem?

Best regards,

jovirus, 2018-04-03:
Hi, can you try nrf connect setting page -> DFU-> allow rename turn on.

avhtak, 2018-04-04:
I turned on the "Allow Rename" function, but DFU has not yet succeeded.
Does Secure buttoless DFU succeed in your environment?

Best regards,

jovirus, 2018-04-04:
can you provide version of SDK you are testing? might be the zip file too, you can send it within the nRF connect app ios

avhtak, 2018-04-04:
I am using SDK12.3.0 + S130v2.0.1 for nRF51822.

I attached secure buttonless DFU files.
1.	Programming “BL_nrf51422_xxac_s130.hex” (bootloader)
Device name will be “DfuTarg”
2.	buttonlessapp_dfu_test0.zip (with nRF Connect for iOS)
Device name will be “Nordic_buttonless”
3.	dfu_tes.zip (for application update with nRF Connect for iOS) <- Error occurred

[DFU_test.zip](https://github.com/NordicSemiconductor/IOS-nRF-Connect/files/1875022/DFU_test.zip)

avhtak, 2018-04-12:
Is there any update about this issue?
This seems to have a disconnect problem after clicking the DFU icon.

jovirus, 2018-04-23:
Hi, could you please send us the bootloader hex file?

avhtak, 2018-04-23:
Hi, I attached bootloader hex file.
Is Buttonless Secure DFU successful in your environment(nRFConnect for iOS)?
Can you provide a successful file?

[BL_nrf51422_xxac_s130.zip](https://github.com/NordicSemiconductor/IOS-nRF-Connect/files/1940374/BL_nrf51422_xxac_s130.zip)

jovirus, 2018-04-24:
@avhtak I have tested your example, it works fine, the only thing you need to do is to enable the experimental buttonless dfu for SDK12.0. I will make a update to nRF Connect to make it configurable.
Thanks for reporting this.

closed 2018-04-24 by jovirus