nordicsemi/IOS-nRF-Connect #27: iOS app crashes on 11.3 on opening of the app

Eryk-MAT, 2018-04-12:
Hello,

In our office, we have tried the latest nRF connect app for iOS on 3 different iPhones and iOS 10.3.

1. It always crashes on opening the app. when Bluetooth is turned on the iPhone
2. there are many BLE devices in our office ether

jovirus, 2018-05-04:
Hey, we need to know how to reproduce the issue you described.

wmacchi, 2018-06-07:
Also experiencing an immediate crash on iOS 11.4 using an iPhone SE. Also crashes on an iPhone 6s. There is no much to reproduce: install from App Store, click to start, crash.
If I turn Bluetooth OFF in Settings the app doesn't crash on start, but it does crash as soon as I re-enable Bluetooth from the Control Center.

spk76, 2018-06-08:
I’m having the same problem. The app crashes on my iPhone 6 with 11.4 and on iPad with v11.3.
Also, fwiw I have a beacon app (Phy) that also recently stopped working.

jovirus, 2018-06-12:
Hi,
The crashing might caused by some of nearby devices. To narrow down, you can set the RSSI filter to a closer range.

wmacchi, 2018-06-12:
Yes, that seems to be the case. Unfortunately the crash seems to be happening before the RSSI filtering is applied, so I am not sure how to isolate the specific device. Other apps I tried (LightBlue, BLE Scanner) don't crash while scanning in the same area.

jovirus, 2018-06-15:
If you send me your appleID, I can add you guys into our Testflight program. You can follow up new builds.

wmacchi, 2018-06-15:
Mine is wmacchi  @  abamis.com

Eryk-MAT, 2018-06-18:
@jovirus its difficult for me to say how to reproduce it.

I open the app it crashes, I attached a video of the problem... Let me know how would you like me to diagnose it 

[IMG_0022.zip](https://github.com/NordicSemiconductor/IOS-nRF-Connect/files/2111330/IMG_0022.zip)

Eryk-MAT, 2018-06-18:
I have loaded the app already and made it crash couple of times

iosdev85, 2018-06-20:
Same for me when I launch the app, it crash. 

If I disconnect BT and launch the app it won't crash, after If I turn back on the BT it's work on iPhone 6s but doesn't change anything on iPhone 6+

jovirus, 2018-06-20:
@Eryk-MAT which version on testflight?

Eryk-MAT, 2018-06-20:
@jovirus just updated to 1.8.6 (6) it seems to be stable so far ... I will keep you posted

jovirus, 2018-06-20:
if you give me your apple id, I can include you in our testflight program @iosdev85

mostafaberg, 2018-07-09:
Is the issue still reproducing after the latest update?
if there's no update on this issue during this week I'll be closing the issue as resolved :)

Eryk-MAT, 2018-07-09:
It seems to be working after the update

closed 2018-07-17 by jovirus