nordicsemi/IOS-nRF-Connect #28: Advertisement interval always 1ms

MaartenMeta, 2018-04-17:
When I discover BLE devices the advertisement interval shows always 1ms with an iPad mini and iPhone X.
With an Android device it shows the correct interval.

mostafaberg, 2018-04-17:
Does it only happen on iPad mini and iPhone X ? or all iOS devices ?

jovirus, 2018-04-17:
The iOS nRF Connect takes interval from last received signal. The Android version is taken an average interval value.

MaartenMeta, 2018-04-17:
Also with an iPad (2017), these are the devices I have so that I can test this with.

mostafaberg, 2018-04-17:
Probably it's as @jovirus says, it might be due to a different implementation on iOS than the android counterpart, do you what the actual advertising interval is ? as in, is 1ms a wrong value ?

MaartenMeta, 2018-04-17:
The actual interval is 1 sec. so it is not a little bit off.

jovirus, 2018-04-17:
if you refresh the device list still does it update ?

MaartenMeta, 2018-04-17:
No it still remains 1ms. I saw it just a moment ago go to 2ms for a small period of time.

mostafaberg, 2018-04-17:
We should probably get a look at this as it might be a bug on our side, will keep the issue opened until this has been looked at and fixed or resolved, @jovirus You'll do that ?

jovirus, 2018-04-17:
We will have a check. Thanks for reporting

jovirus, 2018-04-24:
Hi, @MaartenMeta I have registered your report as an issue, the fix will include in next update. However, you will only get a appox interval measurement on phones. If you want to have a accurate result, you might need a bluetooth sniffer.

closed 2018-04-24 by jovirus