nordicsemi/IOS-nRF-Connect #29: RSSI chart straight line

MaartenMeta, 2018-04-17:
When I discover BLE devices the RSSI chart shows sometimes a straight line and after a while starts bending.
I assume that this is because the iOS device does not detect the BLE device and keeps the value.
This should just be like with the Android device that the line stops when it does not receive a packet within a x period of time.

mostafaberg, 2018-04-17:
Hi @MaartenMeta , do you have a screenshot of how it looks like on both sides so we can have some visual reference when we fix this issue ? thanks a lot for your help and sorry for any inconvenience! :)

MaartenMeta, 2018-04-17:
See pictures below: (the android stops sometimes but this is a bug in the phone itself, other android phones do not show this)
IOS
[image]
ANDROID
[image]

closed 2020-07-29 by dinesharjani