nordicsemi/IOS-nRF-Connect #34: Enabling Boot loader code samples/api  for iOS and Android

MMooreGitHub, 2018-07-18:
Hi,

The nRF Connect iOS and Android apps provides detailed instructions and functionality for enabling the device boot loader via the iOS/Android] mobile apps. When running the  nRF Toolbox DFU update app, I noticed that the DFU update does not place the device in boot loader mode prior to running a DFU update. As a result, the DFU update fails with a "Service Not Available" message. 

So, my questions are:

1. Can you please provide links to both iOS and Android Sample Code (swift/java/kotlin) to enable device bootloader? in nRF Connect, you have to

      write to 0xba to enable the boot loader. 

2. Do you have an iOS and Android API for I may reference as we develop our apps?

    Here is a sample API reference example: https://developer.android.com/reference/android/graphics/Color 

Thanks,

jovirus, 2018-07-19:
The DFU lib for iOS and Android is open source, you find the link on our Nordic github repository.

The nrf toolbox is also open source, which can help you in developing Bluetooth apps.

philips77, 2018-07-19:
Both libs do jump to bootloader mode automatically of you have a standard way of doing so. The method how to enable bootloader mode has changed multiple times in different sdk. Both libs support all legacy and the current solution. It should work out of the box.

If you have a custom way of enabling bootloader, you have to implement it yourselves. Neither of our standard methods require with 0xBA.

jovirus, 2018-07-19:
Hi,
Enter bootloader command is 0x01
The code locate at SecureDFUService.Swift. Line26
Android Bluetooth API https://developer.android.com/guide/topics/connectivity/bluetooth
iOS Bluetooth API https://developer.apple.com/bluetooth/

MMooreGitHub, 2018-07-19:
Hi @jovirus 
Line 26 of SecureDFUService.Swift is: [ static internal let UUID = CBUUID(string: "FE59")]
Can you please provide the method name instead?

MMooreGitHub, 2018-07-19:
Hi @philips77,

Do you have an iOS and Android API for I may reference as we develop our apps?

Here is a sample API reference example: https://developer.android.com/reference/android/graphics/Color

jovirus, 2018-07-20:
@MMooreGitHub 
`internal enum ButtonlessDFUOpCode : UInt8 {
    case enterBootloader = 0x01
    case setName         = 0x02
    case responseCode    = 0x20
    
    var code: UInt8 {
        return rawValue
    }
}`
at file ButtonlessDFU.swift

MMooreGitHub, 2018-07-25:
Hi @jovirus,
Thanks for the sample enumerator. So, it turns out we are using SDK version 12.2.
In line 169 of DFUServiceInitiator.swift within the iOSDFULibrary, it's mentioned that this functionality is "experimental" . So, I've stepped through quite a few classes and am still not able to 
set the DFU to bootloader mode. The The nRF Connect iOS and Android apps provides detailed instructions and functionality for enabling the device boot loader via the iOS/Android] mobile apps
so, it works with this app. However, with the nrF Toolbox code you provided when writing 0xba to enable boot-loader, it does not. I would like to initiate a support call with you or another engineer.
Are you available to discuss this tomorrow?

jovirus, 2018-07-26:
@MMooreGitHub 
From SDK14 the Buttonless DFU is no longer experimental.
I want to make sure I understand your meaning here.
You said you have tried our nRF Connect(iOS/Android), it works to flash your firmware.
But when you try nRF Toolbox which doesn't work.
Do you want to report a bug for nRF Toolbox? What is the question?

mostafaberg, 2018-07-26:
Hi @MMooreGitHub !, let's get your issue fixed :)

- You mention having to use `0xBA` to jump into boot loader, can you give us a reference on where this comes from? a code reference from our libraries, apps that mentions `0xBA` would be appreciated.
- You do not need to do any jump to boot loader commands, we support this by default
- Here is the [documentation](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/documentation.md) for the DFU Library

And the exact code you'll need to flash a peripheral is the following:

- First get your firmware file as follows:
```
let selectedFirmware = DFUFirmware(urlToZipFile:url)
// or
let selectedFirmware = DFUFirmware(urlToBinOrHexFile: url, urlToDatFile: datUrl, type: DFUFirmwareType.Application)
```

- And then start the DFU process
```
let initiator = DFUServiceInitiator(centralManager: centralManager, target: peripheral).with(firmware: selectedFirmware)
// Optional:
// initiator.forceDfu = true/false; // default false
// initiator.packetReceiptNotificationParameter = N; // default is 12
initiator.logger = self; // - to get log info
initiator.delegate = self; // - to be informed about current state and errors 
initiator.progressDelegate = self; // - to show progress bar
// initiator.peripheralSelector = ... // the default selector is used
let controller = initiator.start()
```

We can do a 1 on 1 support but that goes through the tech support team and then they'll most probably redirect to us, so I think it's faster to resolve that here on GitHub to save time, you can still opt-in to go through tech support of course, but I think we can help you faster here. I'm in office all day and I'll be working with you on this today:)

MMooreGitHub, 2018-07-26:
Hi @mostafaberg,

I truly appreciate your proactive assistance regarding this matter.
Is there a phone number or private email I may reach you own.
I would like to share the referenced document in private.

Also, at this point, I'm only reviewing/modifying the nRF toolbox code to try to get this working.
Once this POC is working, we will work towards our company's own implementation.

Thanks!

MMooreGitHub, 2018-07-30:
.

MMooreGitHub, 2018-07-30:
.