nordicsemi/IOS-nRF-Connect #37: Scanner title over writing

DG12, 2018-08-06:
When expanding the details for a tag (ruuvi in particular), The words "Complete Local Name"
overlays "Manufacturer Data"

jovirus, 2018-08-06:
Thanks for reporting,
Which iPhone are you using?

mostafaberg, 2018-08-06:
A screenshot would be very helpful too in that case ! :)

DG12, 2018-08-06:
[image]

DG12, 2018-08-06:
Thanks for your support

DG12, 2018-08-06:
Please see issue #38 for addition problems.

mostafaberg, 2018-08-06:
Perfect, I also reproduced that today locally, fix is coming in the next update :) thanks a lot for your feedback!

mostafaberg, 2018-08-06:
By the way, Which version are you using?

DG12, 2018-08-06:
That was close. screen shot from previous(?) version.
However, 1.8.7 exhibits the same issues, with the addition of the fact that the word
"Scanner" is now in an even larger font causing it to require additional (precious) screen area.
With the message "STOP SCANNING" or "SCAN" at the top right, the title scanner is not really even necessary.

DG12, 2018-08-07:
PS I'd be MORE than happy to try your next release as alpha.

mostafaberg, 2018-08-07:
Ofcourse! Can you send me your iTunes email to mostafa.berg@nordicsemi.no ?
We’ll be looking forward to your feedback :)

DG12, 2018-08-10:
Services has similar problem.
Is it necessary to include the title "UUID" anyway? What else could a string that ugly be? :-)
[image]

mostafaberg, 2018-08-10:
Good point, however such a change should be the same on the android and desktop versions too, so it might be a bit harder to resolve.
Ideally the title should either be the known service (Like device information in this case) or the UUID if it’s an unknown service.

We’ll see what can be done there!

> 
> Services has similar problem.
> Is it necessary to include the title "UUID" anyway? What else could a string that ugly be? :-)
>  <https://user-images.githubusercontent.com/2365961/43956090-cd161858-9c70-11e8-959e-f87ee28e506e.jpeg>

Mostafa Berg
Software Engineer, Mobile Applications Group
P: +47 731 04 195 | M: +47 953 64 082

 <http://www.nordicsemi.com/> <http://www.nordicsemi.com/> <http://www.nordicsemi.com/> <http://www.nordicsemi.com/>
 <http://www.nordicsemi.com/>Nordic Semiconductor <http://www.nordicsemi.com/>
Otto Nielsens veg 12, 7052 Trondheim, Norway
www.nordicsemi.com <http://www.nordicsemi.com/>

  <http://www.facebook.com/nordicsemiconductor/?fref=ts>   <http://twitter.com/NordicTweets>   <http://www.youtube.com/user/NordicSemi>
 <http://www.linkedin.com/company/nordic-semiconductor-asa>

 <http://www.linkedin.com/company/nordic-semiconductor-asa> <https://devzone.nordicsemi.com/>

closed 2020-07-29 by dinesharjani