nordicsemi/IOS-nRF-Connect #38: Detail Specific data poorly formatted

DG12, 2018-08-06:
PLEASE revise the formatting of the Specific data: in the details 
0)Use a fixed width font
1)0x. hexadecimal prefix is unnecessary
2)Add a space between each byte
3)Prevent partial byte from "drooping" onto another line:

currently
```
0x05151C7262C53F0
0280004040073F60
00040C7103C6824C
2
```
Much more usable would be
```
05 12 1C 72 26 2C 53 
F0 02 80 00 40 40 07
3F 60 00 04 C7 10 3C
68 24 C2
```

jovirus, 2018-08-06:
Thanks, I agree. 
We will add it into 'todo' list.

DG12, 2018-08-06:
[image]

DG12, 2018-08-06:
Thanks for you support

mostafaberg, 2018-08-06:
Perfect, this will be coming with the next release, thanks for the report :)

mostafaberg, 2018-08-06:
Actually, I'm sorry, can you confirm this on the latest version of nRF Connect? ?I think I've fixed lots of constraints bugs here in the latest release and this shouldn't be happening

DG12, 2018-08-06:
That was close. screen shot from previous(?) version.
However, 1.8.7 exhibits the same issues, with the addition of the fact that the word
"Scanner" is now in an even larger font causing it to require additional (precious) screen area.
With the message "STOP SCANNING" or "SCAN" at the top right, the title scanner is not really even necessary.

closed 2020-07-29 by dinesharjani