nordicsemi/IOS-nRF-Connect #39: App shows "Suspended" and BLE Scanning Disabled

fillybrese, 2018-08-13:
This has been a bug since first install of iOS 11, I believe. Now I'm on iOS 11.4.1 and both nRF Toolbox and nRF Connect are unusable. I am using iPhone 6S. Same device, no problems with nRF apps before this one.

The scanning list will only populate for a few seconds immediately after phone reboot.

Other BLE Scanning apps (i.e. Adafruit Bluefruit) seem to have problems too, but it is intermittent and can be reset by Bluetooth radio power cycle.

Force closing, re-opening does not fix. Force closing all apps does not fix.

I'm happy to share log files and help diagnose.

[image]

Cheers,
Billy

mostafaberg, 2018-08-13:
Interesting find, haven't seen this before, unfortunately we do not have any API access to know what's wrong, so I suggest doing the following:
* Put the phone out of airplane mode
* Turn on bluetooth from the settings app (Not the control center)
* Restart the phone

And then see if the problem persists, if it also happens on other apps as you say, then there's something on your actual device that's disabling it, perhaps some setting or something ? (I'm not aware of any settings to turn off bluetooth, etc.., but there might be something I don't know)

fillybrese, 2018-08-15:
Hi there, thanks for the reply. I did follow the steps you enumerated, and it did not fix. Actually, what seems to have worked was turn to off WiFi. Then, both nRFConnect and nRFToolbox began reliably scanning. I was able to turn WiFi back on, and both apps have continued to scan reliably since then.

Are you able to tell if this is some unintended side effect between WiFi and Bluetooth APIs in the Nordic iOS Apps, or if its internal iOS issue?

I'll also report back here if i notice any more trends if it begins to act up again and if the fix is repeatable.

Thanks,
Billy

mostafaberg, 2018-10-09:
@fillybrese Any updates ? 😄

fillybrese, 2018-10-15:
Hi @mostafaberg i have found that, when this issue appears (which is still semi-frequently), i have recently found that it's always fixed by toggling Bluetooth radio off/on (not from the Control Center, from the Settings App) as you have recommended. I'm not sure why it didn't work as reported a while ago in my previous message, but it does seem to work well, every time, now. Thank you!

mostafaberg, 2018-10-22:
Seems like an issue with the phone you're having on hand, I'll close this as it's too specific and non-reproducible on our side :(, the best is to follow the workaround you found, until then, there's not so much that can be done on our side as it's not related to nRF Connect :)

If you disagree or find more information to prove otherwise, don't hesitate to reopen the ticket :)

closed 2018-10-22 by mostafaberg