nordicsemi/IOS-nRF-Connect #40: Enable string input when writing to characteristic

carlescufi, 2018-09-14:
There are many characteristics that contain UTF-8 formatted text. It would make sense, just like in the Android version of nRF Connect, to be able to input text instead of hex values for those characteristics, especially the ones that contain a User Description descriptor that states that the contents are text.

CC @mostafaberg @jovirus @roshanrajaratnam @philips77 

This feature will enable users in the (ELCE/OpenIoT conference)[https://events.linuxfoundation.org/events/elc-openiot-europe-2018/] to set their name on the upcoming, Nordic-powered, badge using nRF Connect on both iOS and Android. Requested by @jhedberg

jhedberg, 2018-09-17:
Slight correction to the issue description: it's the Characteristic Presentation Format descriptor that the client should look at, and not the User Description descriptor (which is just free format text)

mostafaberg, 2018-09-25:
Both features are now available on the new release `v1.8.8`, should be available on the App Store within an hour :)

Now you may input byte arrays in hex as before and UTF-8 strings.
additionally, Characteristic presentation format is now parsed, and will display UTF-8 and UTF-16 text values properly, all other presentation formats will be presented in hex for the time being until we have more resources to implement the whole feature properly 👍

closed 2018-09-25 by mostafaberg

jhedberg, 2018-09-25:
@mostafaberg thanks, works nicely! There's one behavioral thing that I was expecting but I guess I forgot to mention it explicitly: wouldn't it make sense to have the text input selected as default instead of hex when the presentation format indicates a text-based characteristic?

carlescufi, 2018-09-25:
@mostafaberg much appreciated, thank you!

mostafaberg, 2018-09-25:
@carlescufi no problems! :)
@jhedberg Ofcourse!, that would make sense, but due to the way things are implemented, It would require a bit more modifications to get that working that way, It makes sense and should be implemented that way, but I had to do the minimal possible amount of changes to get that feature out without touching lots of the codebase, but it'll come as a future update for sure :)

jhedberg, 2018-09-25:
@mostafaberg understood, thanks for explaining.