nordicsemi/IOS-nRF-Connect #41: Tries to Read from a Write descriptor, hangs

bwesen, 2018-09-26:
Hi, I have an u-Blox BLE module I'm developing on (it uses the nrf chips), I'm setting up a custom service ff00 with a custom Read/Notify descriptor ff01 and a Write descriptor ff02 on it.

From the Android version of nRF Connect, everything works as intended.

On the iOS version, as soon as I click on the Service, it seems it tries to READ from the ff02 Write-only descriptor, because I get an unsolicited Read note on the BLE module (it has a built-in ublox connectivity software with AT commands and stuff).

Then the nRF Connect app times out after a while and disconnects.

Bug in the app or in the BLE module?

Bug in my brain? (do you always need a READ permission on descriptors that are going to be used for WRITE only?)

philips77, 2018-09-26:
Your uBox should respond with an error that the desc is write-only (afair). nRF Connect can't know if a descriptor support read or write or both. There are no properties for descriptors.

bwesen, 2018-09-26:
> Your uBox should respond with an error that the desc is write-only (afair). nRF Connect can't know if a descriptor support read or write or both. There are no properties for descriptors.

It shows the properties (READ, WRITE, WRITE RESPONSE, NOTIFY) for each characteristic in the service (my terminology was bugged above maybe, I meant characteristics when I wrote descriptors - I really hate the nomenclature in BLE :).

I installed the BlueSee iOS app instead and that works as intended (doesn't try to read from all descriptors and I can read, notify and write to the descriptors in the service).

I don't think the u-blox software is bug-free, far from it, but it just seems the nrf app also could avoid doing these reads in the first place :)

philips77, 2018-09-26:
Well... I've fixed your comments :)
So yes, nRF Connect shouldn't read from characteristics without Read property, at least not automatically. That's a big on our side, @mostafaberg. But still, i think that the device should respond with some error. It perhaps it does, but nRF just waits for a valid response?

mostafaberg, 2018-10-09:
Strange that it Hangs indeed, this shouldn't be happening, are you stil having the issue or did you manage to resolve it ?

bwesen, 2018-10-09:
I got my iOS custom code working, to access my device with, which worked fine for both read and write to my custom service and characteristics, so I didn't really try the nRF Connect app again in this case.

I think the problem was that the nRF Connect app tries to read from the write characteristic, and then the u-blox firmware for some reason doesn't reject this but instead asks the local MCU for data (the BLE module in this case is connected to the MCU by a serial port doing AT-style commands, and when a read comes in on a characteristic without a value set, it does an unsolicited AT response to ask the MCU for the value). I don't know enough about BLE link-level handshakes to say anything about if it's even possible for a BLE Peripheral to actually allow a read from a char without the read property but it seems like it :) 

So maybe a combination of two bugs :)

mostafaberg, 2018-10-09:
Yeah I understand, most important is that you have things working now, it's most probably a bug on the firmware side since reading from a write only characteristic never hangs any of the chips we worked on :)

closing this for now and if you encounter any issues just reopen or create a new ticket and we'll gladly help 👍

bwesen, 2018-10-09:
To be fair, it didn't hang the chip, it hung the nRF Connect app. The BLE module kept working fine all the time.

mostafaberg, 2018-10-09:
oh I see, I think I misunderstood your bug report then!, did it crash the app or fully hang ?

bwesen, 2018-10-09:
It hung the app, probably because it was waiting for a reply on the read, which the BLE module didn't provide (as I didn't tell it to due to it being a write-only characteristic). I guess it's a matter of definitions, what is supposed to happen in that case I mean :)

mostafaberg, 2018-10-09:
hmmm ok, I'll leave the issue open and try to reproduce and see if I can, then I'll post my findings later on ! thanks again for the feedback, most importantly you got your part working 👍

closed 2020-07-29 by dinesharjani