nordicsemi/IOS-nRF-Connect #44: DFU

HongCongZhi, 2018-10-08:
I can use this app to upgrade with DFU. Not with IOS-nRF-Toolbox. Can you disclose this part of the code? I want to know the use of DFU.

mostafaberg, 2018-10-08:
Such a general question should be posted in our [DevZone](http://devzone.nordicsemi.com) instead.

to answer, yes you can use this app to perform DFU, and no we can't disclose the code since it's not open sourced, but the library is, you'll get better support by asking on DevZone instead of here 👍

closed 2018-10-08 by mostafaberg

HongCongZhi, 2018-10-09:
I used DFU in IOS-nRF-Toolbox and prompted "Scanning for the DFU Bootloader ...", it didn't respond.

mostafaberg, 2018-10-09:
That's also something that the DevZone community can surely help with :) there are lots of reasons this could happen and none of them are related to the nRF Connect app :)