nordicsemi/IOS-nRF-Connect #46: Format value based on characteristic presentation format

0ge, 2018-10-09:
We have a characteristic that has a presentation format set as in the image below. As seen, the presentation format is set to `08 00 00 27 01 00 00`. If my understanding is correct, it should mean that it should be formatted as a UINT32 (which it is). However, when viewing the value its displayed in hex (`0xB9890100`). From what I've [seen](https://devzone.nordicsemi.com/f/nordic-q-a/31253/is-it-possible-to-display-integer-on-nrf-connect-app) this is displayed correctly on the Android version. Will this be supported on the iOS version as well? (Also curious about the desktop version)

[image]

mostafaberg, 2018-10-09:
unfortunately this feature is only available on nRF Connect for Android, nRF Connect for iOS supports UTF8 and UTF16 text representation, other values are displayed as raw data at the moment. it is not planned yet but I'm sure it has to be added at some point in the future :) sorry for the inconvenience

0ge, 2018-10-09:
Happy to see support for it in the future. It really helps when developing the firmware side.

mostafaberg, 2018-10-09:
Yeah that's totally understandable!, yo'll surely be notified if we do anything on that issue from now on, until then you can find the latest and greatest feature on android thanks to super @philips77 🚀  thanks for using Nordic :)

closed 2020-07-29 by dinesharjani