nordicsemi/IOS-nRF-Connect #48: Parse advertised Bluetooth GATT Services data for Health Thermometer and Battery Service

nickgabr, 2019-01-07:
This is an enhancement request for consideration, to enable consistent "user experience" in our BLE sensor quick prototyping/demo development work, by matching an available feature in the app on the Android platform to automatically parse some standard GATT Services within nRF Connect.

## Desired behavior

Similar to how the companion Android app behaves, please recognize the [standard GATT Services](https://www.bluetooth.com/specifications/gatt/services) at least for **Health Thermometer** (`0x1809`) and **Battery Service** (`0x180F`) within an advertising packet, and calculate & display the corresponding data into a temperature in °C and battery level 0-100%.

E.g., as in this screenshot from the companion Android app:

[image]

*This is somewhat related to Issue #42 about the differences between iOS and Android nRF Connect apps, as a specific ask for a feature.*

*Sidenote: the old Nordic Semi demo project [nrf51-ble-app-temp](https://github.com/NordicPlayground/nrf51-ble-app-temp) advertises data in this fashion too, and the Android app [nRF Temp 2.0 for BLE](https://play.google.com/store/apps/details?id=com.nordic.android.nrftemp&hl=en_US) displays and logs the results.  (Which is also a former iOS app, but [the "nRF Temp" app for iOS is now archived](https://appcrawlr.com/ios/nrf-temp), no longer in the App Store.)*

## Existing behavior

The advertised GATT Service UUIDs are listed along with the corresponding data, all in hex.

[image]

nickgabr, 2019-01-07:
I should have given app and platform info for reference:

### Android

App version: 4.22.2 (*latest as of today from Google Play*)
OS version: 9 (*build version PQ1A.181205.006*)
Hardware: Google Pixel 3

### iOS

App version: 1.8.8 (*latest as of today from App Store*)
OS version: iOS 11.4.1 (*15G77*)
Hardware: iPad Mini 2

dinesharjani, 2020-08-14:
We've implemented both Temperature and Battery Level parsing. We're working on adding some more, too.

These will ship in the next version of nRF Connect, available now on our TestFlight beta. If you're interested in trying this out, let us know.

closed 2020-08-14 by dinesharjani

Alebat, 2021-05-07:
Is there a list of formatted characteristics?
I see that 'Temperature' 0x2a6e and 'Humidity' 0x2a6f are shown in hexadecimal format.