nordicsemi/IOS-nRF-Connect #56: DFU aborted on iOS 12.4

EmmanuelBonnet38, 2019-04-30:
Hello,

Randomly DFU is aborted (see attached logs), with iOS 12.X mainly
[Dfu_log_20190416.txt](https://github.com/NordicSemiconductor/IOS-nRF-Connect/files/3130582/Dfu_log_20190416.txt)

We have no issue for DFU with iOS 11.x or android. SDK used is 15.2.0. 

Any hints ?

philips77, 2019-04-30:
Looks like some packets were lost. Try enabling PRN in the DfuServiceInitiator. This will slow down the upload time, but the remote device will have more time to save received data.

bartcerneels, 2019-04-30:
Were getting 308 errors similar to https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues/205

I'm making sure we are running v4.2.2 first.

bartcerneels, 2019-04-30:
> I'm making sure we are running v4.4.2 first.

We are using v4.4.1 and have already tested v4.4.2 without any difference.
The issue I linked to was already fixed in v4.2.0.

EmmanuelBonnet38, 2019-05-02:
Our PRN was 6 and we reduce it to 1. Now it is stable:
self.initiator.packetReceiptNotificationParameter = 1; (default is 12 was 6 before).

Anyway we still don't understand the root cause: 
- why it was always failing at the first packet ?
- why we after an aborted test, when we retry it's always ok.

any hint ?

philips77, 2019-05-02:
I think some packets could have been dropped when new ones were received too quickly. Perhaps they can't be handled that fast. Also, The MTU is 247, so each packet is 244 bytes long and actually more than one packet can be sent in a single connection interval. That's a lot of data to be processed very quickly. Slowing down upload by setting PRN to a small number ensures that both sides are in sync.

I can think of 2 reasons: 
1. As the data are sent as write without response, they are not guaranteed to be received by the application on the DFU target. They may be discarded before reaching the bootloader by the soft device.
2. The flash save operation requires some time and with the data coming too fast it may fail. 

I don't know where exactly the packets are dropped. For sure they are, as the phone has sent 4096 bytes in 244-byte long packets, while the target reported it received only 3852, and not it the right order, as CRC is invalid. In the second retry, it sent 12 packets, each 244 bytes long (2928 bytes), but the PRN received had offset 0, which means that the target didn't get any of the packets. The CRC is not validated for PRNs, only the offset.

And I don't know why the retry works. Perhaps the flash is already erased and no additional time is needed, so flash operations are faster? You would have to ask on DevZone, as this is outside of my scope.

QuentinFarizon, 2020-03-24:
Hello,

I'm seing the same issue on iOS 13.3.1, same messages than @EmmanuelBonnet38 and stuck on "scanning for the DFU bootloader" 
DFU works fine on Android with same zip and same nrf52811 board.
NRF Connect version is 2.2, build 205 (freshly updated)
I have tested with PRN set to 1 in NRF Connect, no changes. Do I have to change something on the bootloader too ?

[image]

Thanks !

QuentinFarizon, 2020-04-03:
Hello,
I can confirm it works with disabling "Enable Alternative Advertising mode"

What would need to be changed in the bootloader or app to make this alternative advertising work ?

philips77, 2020-04-04:
Add you can see in the log on there photo, the app sends a new name, Dfu70999, by sending command 0x02... The device should reply with 0x60-02-01 (response code-set new name-success). Then the app should send 0x01 to restart the device into bootloader mode.
In your case, the device just restarted, which looks like a big on the bootloader. It's also a bug in the dfu lib, it should not assume the name was set correctly. I'll fix it on Monday.

QuentinFarizon, 2020-04-04:
@philips77 Thanks a lot for that !
I will look at the root cause on my end (why setting the name fail). Where should I start looking in your opinion ?

philips77, 2020-04-05:
I would recommend asking on Nordic DevZone. Our support team will know such things. Also they can report it to the devs, if an issue is found.

QuentinFarizon, 2020-04-05:
@philips77 You're right, done, here it is : https://devzone.nordicsemi.com/f/nordic-q-a/59881/advertising-rename-feature-not-working

philips77, 2020-04-06:
Did the fix suggested fixed your problem? I could print a message for debugging in DFU if such problem is detected.

QuentinFarizon, 2020-04-06:
Hello @philips77. Yes, suggestion on the ticket solved the issue ! Indeed an issue in my code.

It would be nice to also fix this on iOS app as you suggested. When it happens, the DFU hang as previsouly stated, but an other issue is that even when cancelling, NRF connect indicates it is disconnected, but the device is still connected in Bluetooth I think, as I cannot connect from another device, so it needs to be restarted.

philips77, 2020-04-06:
Ok.. so the device didn't reboot, but rather froze. That means, that the only thing I can do in the DFU lib is to print a hint how to solve this problem. Still better than nothing. Thanks for your comment.

philips77, 2020-04-08:
The issue has been "fixed" in the DFU library 4.7.1. Now, you'll see the link to this DevZone case if such issue happen again.

closed 2020-04-08 by philips77