nordicsemi/IOS-nRF-Connect #59: DFU Freezes

jaredwolff, 2019-07-16:
**What happens**

When I go through this process, I can reliably get the DFU process to lock up:

1. Open app
2. Scan for device
3. Connect to device
4. Go to DFU screen
5. Select file
6. Hit the play button

But, when I disconnect and follow these steps it works

1. Hit the DFU button (need to be in the device screen but disconnected)
2. Select DFU file
3. Then hit the play button.

If it does lock up, I have to disconnect, start pairing and then follow the second set of steps. 

To summarize: when a DFU is initiated when already connected it almost always locks up. When DFU is initiated when disconnected, the device will connect and successfully update the device.

**What should happen**

In the first case, the DFU should continue as normal and update the device.

jaredwolff, 2019-07-16:
I'm running SDK 15.2 and version 1.8.8 of nRF Connect on iOS

ihpang, 2019-07-16:
Hi,

I also face the same issue with nRF Connect IOS freeze when running SDK 15.2. This problem doesn't happen with Android version of nRF Connect...

ihpang, 2019-07-17:
> I'm running SDK 15.2 and version 1.8.8 of nRF Connect on iOS

Hi There,

Did you managed to get any solution?

jaredwolff, 2019-07-17:
@ihpang there is a workaround. It's a hack but it works every time.

1. Connect to your device as normal. 
1. In the connection screen, **disconnect**
1. Even while disconnected you can still press the **blue DFU button**. Click it.
1. Select your image
1. If you device is not advertising, start advertising again.
1. Finally hit the blue play button. The app will attempt to connect and then update. 

Seems like it happens only when the device is **already connected**. If it's not connected and then connects to do a DFU it's fine. (for buttonless) If your device is already in DFU mode I believe it works either way.

jaredwolff, 2019-07-17:
@ihpang the alternative also is to use NRF Toolbox. Just tried it and it seems to be working ok. It appears that it doesn't connect first, then try to DFU. It connects and puts the device into DFU immediately.

[I also opened a thread on Devzone.](https://devzone.nordicsemi.com/f/nordic-q-a/49829/nrf-connect-dfu-update-not-proceeding-on-pre-connected-device-ios-app)

dinesharjani, 2019-08-02:
Sorry about the issue - we're nearing completion of a brand-new version of nRF Connect, which of course, includes support for DFU Updates.

[Email me](mailto:dinesh.harjani@nordicsemi.no) if you'd like to beta-test it. We believe the DFU update functionality works a bit better in 2.0.

closed 2020-07-29 by dinesharjani