nordicsemi/IOS-nRF-Connect #65: Feature request: Enable copy from characteristic fields

andresovela, 2019-11-19:
Hi,

I would like to request a feature so that it is possible to copy the values read from a characteristic. See screenshot for context.
[image]

I think it would be nice to be able to copy that long value in order to be able to use it somewhere. At the moment I'm forced to manually type the values (just for debugging purposes).

andresovela, 2020-01-16:
Any updates on this? :( I would _really_ appreciate this feature.

dinesharjani, 2020-08-10:
[image]

Implemented. Closing the issue. This will ship in the next release of nRF Connect for iOS.

closed 2020-08-10 by dinesharjani

andresovela, 2020-08-10:
Nice!