nordicsemi/IOS-nRF-Connect #67: DFU does not work on iOS 13.4

istvanns, 2020-03-30:
Hi! DFU hangs and seemingly it doesn't even start. DFU works both with the nRF Connect desktop and Android apps.

iOS version: 13.4
nRF Connect version: 2.2 build 205

[image]

philips77, 2020-03-30:
Hello,
We need some info from you:
1. How are you?
2. What SDK is your bootloader based on?
3. Can you share logs? Where does the DFU stop?
4. Did you click Start button? It says "Ready to Upload".

istvanns, 2020-03-31:
Hi,

Thanks for you reply! Let's see the points:

1. I'm fine, thanks! How are you?
2. SDK 16.0.0
3. DFU doesn't even start. I don't see any option to get logs in the iPhone app.
4. Tapping the "Start" button does nothing in my case. In an other case, we get stuck at "Scanning for the DFU Bootloader...", which is like issue #56

philips77, 2020-03-31:
You may try disabling the "naming feature" in app settings -> dfu -> Enable Alternative Advertising Mode. For some people it doesn't work.
The app, at this moment, needs to switch your device to bootloader mode. It sends a command "0x01" to the buttonless characteristic, which should cause your device to restart. If the renaming feature is supported (SDK 14+), before sending 0x01, the library would send 0x02 followed by a random name. The bootloader would then advertise with this random name, so the lib could reconnect to it (MAC address isn't available on iOS). Bot for some clients for some unknown for us reasons, this fails, so there's a switch to disable it.
And you have to click Start. Perhaps it's small and difficult to click?

istvanns, 2020-03-31:
The "naming feature" isn't the problem in my case. The "Start" button simply doesn't work in portrait mode. I rotated the device and the button works only in landscape mode, although the UI is messed up (see screenshot). The DFU update then works, even with the "naming feature" enabled. The point is, the "Start" button does not work in portrait mode on the iPhone 8 (no matter how and where I tap it). The problem is not the size, since it works in landscape mode.

Device: iPhone 8
iOS version: 13.4
nRF Connect version: 2.2 build 205
Bootloader SDK version: 16.0.0

[image]

dinesharjani, 2020-03-31:
@istvanns Thank you very much for bringing up this issue. We're going to make UI changes this week to give the layout a lot more room to breathe.

I'd be very helpful if you could give me an email to add you to our TestFlight group so you can test it ahead of everyone else. How about it?

istvanns, 2020-06-26:
Then new version solved it. Thanks!

closed 2020-06-26 by istvanns