nordicsemi/IOS-nRF-Connect #69: Changing advertising data is not updated when scanning (but works fine on Android)

jeremyherbert, 2020-05-07:
I have an application very similar to the `scan_adv` example in the zephyr project. Essentially, I have a device with a static name and static address which starts and stops advertising every second, and advertises the data 0xFF 0xFF 0xNN where NN is a number that increments every second.

In the android nrfconnect app one can see the advertising data update every second (as long as it is still scanning). However, in the ios app it just reads the advertising data once and does not update it even if scanning is stopped and then started again (by tapping the button in the top right). The only way to get it to update is to do the swipe down gesture to clear the list of scanned objects.

dinesharjani, 2020-07-29:
Hi @jeremyherbert - is this still happening with the current version of nRF Connect in the App Store?

D0m0, 2021-06-05:
I have the same problem, it indeed only appears on iOS (including the latest version of the app). So I tried everything from clearing application data to reinstalling. And suddenly it occurred to me that maybe it was not the app, but the iOS bluetooth stack which cached this data? I switched to airplane mode and back, and it was all gone!
Hope this helps!

ryanwallban, 2021-10-21:
-

pedson, 2023-05-31:
Has this been confirmed to be an issue w/iOS caching the advertisement data and you cannot override/forcibly request an update?

I've noticed that if the app does a full reset (launches with filters cleared, etc, which does not happen after every force quit - not sure what triggers it) it also refreshes the data.  You don't always have to toggle BT on and off.  Regardless 99% of the time that you relaunch the app it doesn't refresh.

This is a fairly debilitating behavior when working with advertising data.

dinesharjani, 2023-06-01:
@pedson trust me, I tried. Even on the Mac this is not easy to do, because the IOBluetooth API is for Classic Bluetooth and the CoreBluetooth APIs on the Mac are basically the iOS ones.

If you find a way to get the BLE cache cleared, let me know. I'm interested :)