nordicsemi/IOS-nRF-Connect #70: Going to landscape crash on iOS 13.5.1

mhammerc, 2020-06-07:
Hello!

I have been using this application for years and it is so useful and great 😋!

On iOS 13.5.1 on iPhone 11 Pro, when going to landscape the app crash, only when I am in a device page.

This is a low priority bug of course.

[image]

dinesharjani, 2020-06-08:
Hi Martin, 
Thank you for your feedback! I'll try my best to reproduce this now.
I have a request though: do you mind attempting to reproduce this issue on your device using our TestFlight beta of version 2.3? We're pretty far in development, and we could use your experience with it.

All that we require is your authorization and that you provide us an email to send you the invite to.

Thanks!

mhammerc, 2020-06-08:
Hello!

Thank you for your answer!

Of course I accept.

How can I give you my mail address elsewhere than a public place?

dinesharjani, 2020-06-08:
> Hello!
> 
> Thank you for your answer!
> 
> Of course I accept.
> 
> How can I give you my mail address elsewhere than a public place?

Report an issue inside nRF Connect :)

mhammerc, 2020-06-08:
Done :)

closed 2020-07-31 by dinesharjani