nordicsemi/IOS-nRF-Connect #72: DFU service not recognised

istvanns, 2020-06-25:
[image]

iOS version: 13.5.1
App version: 2.3 build 306
Bootloader SDK version: 16.0.0

DFU works with the Android app version 4.24.1 on Android 10 with the same device, so it must be something with the iOS app.

philips77, 2020-06-26:
Please, restart Bluetooth in iOS settings and try again. Perhaps some previous services have been cached by the system and restarting Bluetooth will fix this.

istvanns, 2020-06-26:
Right. Had to restart a few times, but it solved the problem. Also had to turn off alternative advertising name (or something similar). Thanks!

closed 2020-06-26 by istvanns