nordicsemi/IOS-nRF-Connect #74: Log is empty

serotta58, 2020-07-21:
On version 2.3.1 on an iPad, when I swipe to view the log for a connected device, the log text is empty regardless of what I set the filters to, and the plaintext / CSV format option.  Pressing the Share button flashes the possible sharing options, but then closes right away, presumably because there is nothing in the log.

dinesharjani, 2020-07-29:
Thank you @serotta58 - we're making changes to the app so this can't happen.

This should be fixed in the next release. We're currently not planning any bug-fix releases in the 2.3.x line, so this might take a bit.

dinesharjani, 2020-07-31:
This is fixed and will ship in the next version of nRF Connect.

closed 2020-07-31 by dinesharjani