nordicsemi/IOS-nRF-Connect #75: Attribute Table Not Updating

akash-patel, 2020-07-30:
iOS 13.6
iPhone 11 Pro
nRF Connect v2.3.2

Steps to reproduce using nRF52840 DK:

1. Flash application (ble_app_buttonless_dfu_without_bonds_s140_with_setting_page.hex), bootloader (bootloader_secure_ble_debug_without_bonds_s140.hex), and softdevice (softdevice_s140.hex). These files are from the nRF5_SDK_17.0.0_9d13099\examples\dfu\secure_dfu_test_images\ble\nrf52840 and inside the .zip attachment below for convenience. This can be flashed using the Programmer tool inside of nRF for Desktop. 

2. Perform Buttonless DFU of application using hrs_application_s140.zip. This is also in the same location in the SDK as above and inside the .zip attachment below.

3. Now device should be advertising as the heart rate application. Erase device using nrfjprog -e and reprogram application, bootloader, and softdevice as in step 1.

4. Now device should be advertising as Nordic_Buttonless. However, when you connect to it, the attribute table still shows the services/attributes from the heart rate application and therefore we cannot do a buttonless DFU. There is a video showing this issue in the attachment below.

All files and video of issue can be found here: [Files_and_Video.zip](https://github.com/NordicSemiconductor/IOS-nRF-Connect/files/5003738/Files_and_Video.zip)

akash-patel, 2020-07-30:
Meanwhile when using nRF Connect for Desktop BLE tool, the Secure DFU service is there.
[image]

dinesharjani, 2020-07-31:
Hi @akash-patel ,

It is likely you need to refresh iOS' cache of the services provided by said Peripheral. You need to go to Settings and turn off, then on, Bluetooth. Don't use Control Center as that switch doesn't turn off Bluetooth - it merely 'silences' it, so there are no events.

If indeed this is a iOS-cache-problem, we cannot do anything, as we merely provide in the app what iOS' CoreBluetooth API gives us.

mhammerc, 2020-08-03:
On a development workflow, this is a real issue for inexperienced users.

Many times I had this issue on Linux (w/ Bluez) never updating its cache, requiring to disable cache in Bluez settings.

I don't recall it happened on nRF Connect for iOS in my workflow, maybe because I have not been pairing my devices.

If this it's true it happen on iOS, maybe insert in nRF Connect a button or something which explain how to refresh cache as you just said? (So the user can learn what's going on)

philips77, 2020-08-04:
Good suggestion. @dinesharjani how about a button somewhere explaining that user needs to reset Bluetooth in settings?

akash-patel, 2020-08-04:
@dinesharjani, toggling Bluetooth in Settings resolved the issue but as the others have mentioned above, it would be good to have this explained in the app in case others come across the same issue.

dinesharjani, 2020-08-10:
I like the idea - it could also clean-up the Settings Section a bit. I'll work on a new "FAQ" page in Settings that explains this issue.

dinesharjani, 2020-08-10:
Will report-back here once it's implemented.

OevreFlataeker, 2020-08-17:
I have the same issue but I am pretty sure it worked a couple of days ago and the attributes updated correctly w/o any need for toggling Bluetooth on the phone. The described work around works here as well.

closed 2024-03-11 by dinesharjani