nordicsemi/IOS-nRF-Connect #76: Cannot send current time to peripheral

albsod, 2020-08-08:
With the Android version of nRF-Connect, it's possible to add a current time profile, connect to a peripheral and send the current time to it. With the iOS version, this doesn't work.

I've tried to add a number of services manually under Advertiser:

- Generic Access
- Generic Attribute
- Current Time Service
- Reverence Time Update Service
- Heart Rate

This allows me to send some data it seems, but the time and date components will be zero.

philips77, 2020-08-08:
Advertiser and gatt server are two different things, not much related. We are working on gatt server, but it may take some time until it's done. It's a pretty big feature.
But as far as I remember, iOS contains this service by default. Perhaps it requires bonding. Just connect to advertising nrf connect for iOS and discover services.

jaiminajmeri, 2020-12-16:
@albsod I recently started working with the Current Time Service client application. I tried importing `cts_central.cts` file in the mobile app but it gave an error. I think It's supported by the desktop version.

I created the XML version of the cts_central by comparing it with the CTS specification and it worked. I configured the GATT Server by importing the XML file. I have shared the file here.
https://github.com/jaiminajmeri/ble-services-nrf-connect/blob/main/Current_Time_Service_v1.1.0.xml

dinesharjani, 2020-12-16:
@jaiminajmeri so you're using the new GATT Server feature in nRF Connect 2.4.x and it works? 
In other words, we can close the issue?

jaiminajmeri, 2020-12-16:
@dinesharjani I am using the Android version of nRF Connect 4.24.3 and I have tested it by importing the XML file that I shared, worked for me. Not sure if this means to close the issue.

closed 2021-01-07 by dinesharjani

philips77, 2021-01-19:
@jaiminajmeri did you try the same with nRF Connect for iOS, version 2.4+?

jaiminajmeri, 2021-01-21:
@philips77 

I have an iPad Mini 5 running on iPad OS 14.3 (18C66). I updated nRF Connect to 2.4.7, tried using the `Peripheral->SERVERS` section.

I am testing the `Current Time Service`. First, I tried importing an XML file by using the `Import XML Server File` button. It showed the message below. Then later I tried creating the server manually using the `Add Server` button, got the same message.

Clicking the slide switch shows this message:
```
Unable to Advertise
One or more parameters were invalid.
```

Am I missing something?

philips77, 2021-01-21:
iOS is handling some services natively. For example Battery Service, Current Time Service, ANCS, etc. Those cannot be added by apps to the peripheral manager. If you try a different service, like Nordic LED and Button Service, that should work.
To test Current Time Service you don't need to do anything, as you always get this service on a connected BLE device and it is the iOS that provides the time. On Android there's no such feature, so it is possible to add it in nRF Connect for Android and send any data from it.

The only issue here is that you get somewhat misleading error. We will fix that in the next revision, and will remove services that generate such error.

jaiminajmeri, 2021-01-22:
@philips77  thank you! for explaining what's expected from the iOS version of the app, I have some clarity now.