nordicsemi/IOS-nRF-Connect #77: Scanner sometimes partially identifies Ruuuvi tags

DG12, 2020-08-28:
Sometimes Ruuvi Innovations Ltd tags (mfg ID 0499) are identified but sometimes they show N/A (even with Ruuvi logo!)
See attached.
Sometimes Ruuvi tags show Ruuvi icon, but sometimes show DB9 - Nordic UART Service.
Continues with version 2.4 build 136
[image]

dinesharjani, 2020-09-02:
Hi @DG12 ,

I see only one device that doesn't show the advertised name for the device, could it be because that device in particular simply doesn't advertise a name? All of the others seem to show up correctly in the UI.

DG12, 2020-09-02:
Why does
Ruuvi1E1A correctly show Ruuvi Icon and name but
RuuviEC0F shows UART icon as does Ruuvi20BC

dinesharjani, 2020-11-09:
Looking at the screenshot, my theory would be that Service Data has higher priority than Manufacturer Data, and the scans that don't say "Ruuvi" show some Service as being Advertised.

If you clear the BLE data (Settings -> Bluetooth -> Turn Off / Turn On, don't use the Control Panel) it might clean up the Scanning Data.

closed 2024-12-12 by dinesharjani