nordicsemi/IOS-nRF-Connect #78: Scan should not Clear RSSI and interval  when Scan stops.

DG12, 2020-09-02:
Scan should not Clear RSSI and interval  when Scan stops. 
The current implementation 2.4#136 shows the values while scan is running and then suddenly, without warning, clears them (when the scan timer runs out).

A better approach would be  show value in white while running, then in gray(silver?)  when scan is not active.

dinesharjani, 2020-11-09:
That is a nice idea.

Early on in development, this was how it was implemented. However, it was deemed that this information "was not correct" because you could be reading the Scanned data hours, maybe even days later when it wouldn't be "meaningful." But we can follow your suggestion for sure. We have our hands full trying to finish 2.4, but if we get to it, we'll implement this.

Thanks for your feedback.

DG12, 2022-11-17:
Seems that this has been implemented . Thank you.

closed 2022-11-17 by DG12