nordicsemi/IOS-nRF-Connect #79: On rotating the iPhone the app hangs, then crashes (2.3.2b59, iOS 13.7)

chrislro2019, 2020-09-22:
I have 2.3.2 build 59 running on iPhone 8 iOS 13.7. The apps works fine when phone is vertical. When rotating the phone to horizontal, the app hangs (showing the rotated but only upper part of the screen), then is killed after some seconds.

dinesharjani, 2020-09-22:
Hi @chrislro2019,

We're working on a big update, version 2.4, that is a bit off from release still. What we can do is we can invite you to become a TestFlight beta tester (just provide us with an email to send you an invitation), and you can confirm to us we've fixed this.

I believe a user already notified us about this issue.

chrislro2019, 2020-09-22:
I installed the test flight version. Thank you for that! Its behavior better, but still crashes after (now) multiple rotations ... I have send a crash report via Test Flight.

chrislro2019, 2020-09-25:
Seems to be fixed in the latest 2.4 TestFlight build (not yet officially available). Thanks!

closed 2020-09-25 by dinesharjani