nordicsemi/IOS-nRF-Connect #81: can't read data in utf-8 form on nrf connect

sm-oralnum, 2020-11-04:
I use nRF connect to read from my application in the raspberry. I am able to set the services and the characteristics. I can read, write and get notifications. The problem is that i get the data in hexadecimal form when i read them using the app (nRF connect). I want to get them in the form of string(utf-8). Does anyone know how to do that?

dinesharjani, 2020-11-09:
Hi @sm-oralnum ,

Unlike the Android version of nRF Connect, on iOS we don't yet allow you to set a parser for teh Data received. We try to detect the data type automagically, but we don't cover many cases. If you give us the specific Service/Characteristic UUID you're trying to work on, we can fix it for you.

sm-oralnum, 2020-11-09:
Hi @dinesharjani , i am writing my own Service/Characteristic. We will say that i write Hello on my Characteristic and want to read the data in utf-8 form. I can read it only in hexa form. Is there a way to read it in utf-8 form from the app?

dinesharjani, 2020-12-10:
@sm-oralnum Not right now. What you can try is to add 'String' to the name of the Characteristic; this should make nRF Connect attempt to parse the Data in UTF-8 form. Let us know if this works.

meralus1990, 2021-02-19:
- - [x] ￼I'm not sure why but I seem to be having the sae

andrewleech, 2021-03-22:
Hi, this problem of only displaying hex makes nrf connect unusable for debugging our device - we have a large MTU so every read just gives a huge wall of hex that's impossible to view.

Please make the app just show both hex and text, like the android one, at least that works for everyone.

Even better, add support for the same `uuid_definitions.json` file that android and windows supports so we can make our own definition file?

Please!

dinesharjani, 2024-03-11:
Marked as closed since nRF Connect has added 'set Parser' feature.

closed 2024-03-11 by dinesharjani