nordicsemi/IOS-nRF-Connect #83: Cannot add Battery Service in Peripheral Simulator

OevreFlataeker, 2020-12-06:
I want to offer the CSCS + Battery Service using the Peripheral simulator function in the new nRF Connect Toolbox for iOS.
Whereas CSCS works, as soon as I add the Battery Service 0x180f and the corresponding characteristic Battery Level 0x2a19, upon saving and starting the server via the "Advertise" button the following error appears:

"Unable to advertise, the specified UUID is not allowed for the operation" 

The error appears if I setup Battery Service in the Advertiser section or not and no matter if I have a CSCS service added or not. As soon as I add a simple Battery Service 0x180f this error appears.

I might be doing something wrong but then, it's not really clear to me who to properly setup the service?

dinesharjani, 2020-12-10:
Hi @OevreFlataeker ,

Sorry for taking so long to get back to you. The error you see has its source in iOS' BLE framework when we tell it to add the set of Attributes to advertise; from our experience, it likes it more when a Server Configuration has Attributes to relate to one thing, so for example, you may try to have one Server Configuration for the Battery Service and another for CSCS. 

That's all we can suggest at the moment.

OevreFlataeker, 2020-12-10:
Thanks dinesharjani. So you're basically saying I cannot advertise multiple services in one server config on iOS? Is this what you mean? Would that work in the Android version?

dinesharjani, 2020-12-10:
I believe Android should me more allowing, yes.

philips77, 2021-01-19:
Hi, @dinesharjani as GATT server is independent from advertising, you should be able to add multiple services to peripheral simulator. You should not start advertising at all. The error you get I believe is caused by insufficient space in the adv packet, and when you remove the dependency, it should work.

closed 2024-12-12 by dinesharjani