nordicsemi/IOS-nRF-Connect #84: App always crashes when stopping advertising and central connected

kacper1703, 2020-12-10:
App: 2.4.4, build 37, iPhone 11 Pro

Repro steps:
- In Advertising Tab, create a new Server
- Add a service - "Legacy DFU Service"
- In the service add a characteristic "Blinky button state", 
-- properties: Notify, Indicate; 
-- permissions: none; 
-- data: Bool (false)
- Start advertising
- Connect a central and subscribe to notifications on "Blinky" characteristic
- (optional) send a value notification to the central
- Tap "Stop Advertising"
- **App crashes**

App does not crash if I disconnect the central on the central side, then stop advertising.

kacper1703, 2020-12-10:
Also, any attempt to open control center or moving the app into the background causes the app to crash as well.

dinesharjani, 2020-12-10:
@kacper1703 Thank you!
We're going to look into this immediately.

dinesharjani, 2020-12-10:
@kacper1703 Your first issue, the crash when tapping on "Stop Advertising", has been fixed, and will be part of the next update, 2.45.

However, I cannot reproduce the background crash. Do you have specific steps to reproduce it as well?

kacper1703, 2020-12-10:
They're the same, but instead of tapping "Stop Advertising" I either move the app to the background or for example try to open Control Center. Locking the phone also crashes the app.

dinesharjani, 2020-12-10:
@kacper1703 both issues were caused by the same bug; it's fixed and committed to the 2.4.5 branch. Thank you!

closed 2020-12-10 by dinesharjani

kacper1703, 2020-12-10:
Awesome, thanke you. When could we expect the 2.4.5 release?

dinesharjani, 2020-12-10:
@kacper1703 I expect it to be out very soon - at the latest, middle of next week. But I think sooner, because we're very close to our Christmas break :)