nordicsemi/IOS-nRF-Connect #85: Working on Android not on iOS

supersunil, 2020-12-28:
I've flashed the bootloader and programmed my Adafruit nRF52840 Express board to perform the OTA DFU operation.

I've opened the example code and compiled it using Arduino 1.8.13 and I've double checked the bootloader version (0.3.2 with softdevice s140 6.1.1) on the board as well as in Arduino IDE.

I've moved the compiled code .ino.zip to my Android Device running Android 8.0.0. I've installed the nRF Connect 4.24.3 and followed the process of OTA Device Firmware Update, it was successful (shown as video in the attached drive link)

https://drive.google.com/file/d/1wC63r85xd41FOQqn4Ciw4HilG8Ex8iqu/view?usp=sharing

Now I'm performing the exact same process on my iPhone running iOS 14.3 and the nRF Connect 2.4.6 but it exits with operation failed status (shown as video in attached drive link)

https://drive.google.com/file/d/1UuRIBfP-XKB9VHR4V0gn2aGhyWrbDUoN/view?usp=sharing

Help me to get rid of it.

philips77, 2020-12-28:
The app your using on the video is nRF Toolbox, not nRF Connect, for which you have created the issue. Please, try the same on nRF Connect for iOS. I believe that issue was fixed in one of the latest version of Dfu library, and the current version of nRF Toolbox may be using an older one.

supersunil, 2020-12-28:
So you completely believe that this is an issue of an iOS application not an issue of OTA DFU library on the device?

supersunil, 2020-12-28:
I've tried the same process in nRF Connect for iOS. Failed this time too. This is the video link.

nRF Connect log:
`11:56:08,DFU,normal,File Name: blinky_ota-nrf52.ino.zip
Parts: 1
Size: 80 KB
Soft Device Size: Zero KB
Bootloader Size: Zero KB
11:56:11,DFU,debug,[Callback] Central Manager did update state to: Powered ON
11:56:11,DFU,normal,Connecting to BLEUART52...
11:56:11,DFU,debug,centralManager.connect(peripheral, options: nil)
11:56:11,DFU,debug,[Callback] Central Manager did connect peripheral
11:56:11,DFU,normal,Connected to BLEUART52
11:56:11,DFU,normal,Discovering services...
11:56:11,DFU,debug,peripheral.discoverServices(nil)
11:56:11,DFU,normal,Services discovered
11:56:11,DFU,normal,Starting Legacy DFU...
11:56:11,DFU,normal,Connected to BLEUART52
11:56:11,DFU,normal,Services discovered
11:56:11,DFU,normal,Legacy DFU Service found
11:56:11,DFU,normal,Discovering characteristics in DFU Service...
11:56:11,DFU,debug,peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
11:56:11,DFU,normal,DFU characteristics discovered
11:56:11,DFU,normal,Reading DFU Version number...
11:56:11,DFU,debug,peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
11:56:11,DFU,normal,Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0100
11:56:11,DFU,normal,Version number read: 0.1
11:56:11,DFU,normal,Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
11:56:11,DFU,debug,peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
11:56:11,DFU,normal,Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
11:56:11,DFU,normal,DFU Control Point notifications enabled
11:56:11,DFU,warning,Application with buttonless update found
11:56:11,DFU,normal,Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
11:56:11,DFU,debug,peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
11:56:11,Scanner,normal,Device Appearance switched from UART to NordicDFU.
11:56:12,DFU,debug,[Callback] Central Manager did disconnect peripheral
11:56:12,DFU,normal,Disconnected by the remote device
11:56:12,DFU,normal,Connecting to BLEUART52...
11:56:12,DFU,debug,centralManager.connect(peripheral, options: nil)
11:56:12,Connected Device,debug,[Callback] centralManager(central, didDisconnectPeripheral: peripheral, error: The connection has timed out unexpectedly.)
11:56:12,Connected Device,normal,Disconnected.
11:56:13,DFU,debug,[Callback] Central Manager did connect peripheral
11:56:13,DFU,normal,Connected to BLEUART52
11:56:13,DFU,normal,Discovering services...
11:56:13,DFU,debug,peripheral.discoverServices([00001530-1212-EFDE-1523-785FEABCD123])
11:56:13,DFU,normal,Services discovered
11:56:13,DFU,normal,Legacy DFU Service found
11:56:13,DFU,normal,Discovering characteristics in DFU Service...
11:56:13,DFU,debug,peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
11:56:13,DFU,normal,DFU characteristics discovered
11:56:13,DFU,normal,Reading DFU Version number...
11:56:13,DFU,debug,peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
11:56:13,DFU,normal,Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0800
11:56:13,DFU,normal,Version number read: 0.8
11:56:13,DFU,normal,Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
11:56:13,DFU,debug,peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
11:56:13,DFU,normal,Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
11:56:13,DFU,normal,DFU Control Point notifications enabled
11:56:13,DFU,normal,Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
11:56:13,DFU,debug,peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
11:56:13,DFU,normal,Writing image sizes (0b, 0b, 79736b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
11:56:13,DFU,debug,peripheral.writeValue(0x000000000000000078370100, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
11:56:14,DFU,normal,Data written to 00001531-1212-EFDE-1523-785FEABCD123
11:56:14,DFU,normal,Start DFU (Op Code = 1, Upload Mode = 4) request sent
11:56:17,DFU,normal,Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100101
11:56:17,DFU,normal,Response (Op Code = 1, Status = 1) received
11:56:17,DFU,normal,Writing Initialize DFU Parameters...
11:56:17,DFU,normal,Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
11:56:17,DFU,debug,peripheral.writeValue(0x0200, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
11:56:17,DFU,normal,Writing to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
[... 2 lines cut ...]
11:56:17,DFU,debug,peripheral.writeValue(0x0201, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
11:56:17,DFU,normal,Data written to 00001531-1212-EFDE-1523-785FEABCD123
11:56:17,DFU,normal,Data written to 00001531-1212-EFDE-1523-785FEABCD123
11:56:17,DFU,normal,Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100201
11:56:17,DFU,normal,Response (Op Code = 2, Status = 1) received
11:56:17,DFU,normal,Initialize DFU Parameters completed
11:56:17,DFU,normal,Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
11:56:17,DFU,debug,peripheral.writeValue(0x080a00, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
11:56:17,DFU,normal,Data written to 00001531-1212-EFDE-1523-785FEABCD123
[... cut ...]
11:56:18,DFU,normal,Response (Op Code = 3, Status = 6) received
11:56:18,DFU,error,Error 6: Operation failed
11:56:18,DFU,normal,Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
11:56:18,DFU,debug,peripheral.writeValue(0x06, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
11:56:18,DFU,debug,[Callback] Central Manager did disconnect peripheral
11:56:18,DFU,normal,Disconnected by the remote device`

Video Link:
https://drive.google.com/file/d/1OZffcZcRcYrz3VltxZy1f6vwZrICTfTu/view?usp=sharing

dinesharjani, 2021-01-07:
Looks like a DFU Library issue.

philips77, 2021-01-19:
@dinesharjani to what value did you set https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/d4a1da27fbf195b325cea3036235a4939a8f4efa/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L226 in nRF Connect for iOS? The Android version has 0.3 or 0.4 sec.

dinesharjani, 2021-01-19:
@philips77 on iOS it's not set - it remains on the default value.

philips77, 2021-01-19:
If you could add that to settings, set to 0.3sec by default.

dinesharjani, 2021-01-20:
This has been added into the new 2.4.8 branch and will ship with it.

Closing this issue.

closed 2021-01-20 by dinesharjani