nordicsemi/IOS-nRF-Connect #89: New server - Configuration

mediaformat, 2021-03-03:
```
App version: 2.4.8
Build number: 163
iPhone 6s, iOS 14.4
```
When configuring a New Server.

Adding a Service always adds the following services in the following order:

1. User Data, 
2. Reconnection Configuration, 
3. Scan Parameters, 
4. Immediate Alert
5. Internet Protocol Suport Service

Adding a new Service after the 5th, a Duplicate Attribute error is triggered.
Though the contextual menu for each Service features an Edit option, it does nothing.

This seems buggy. My, albeit n00b, thinking is that Adding a New Service should prompt picking a Service Type rather than just enumerating the mentioned List. (For example Current Time Service). 
And the Edit option should allow Modifying a Service Type. 

The behaviour is similar when adding Characteristics and Descriptors: Default names are unmodifiable. So unusable?!

itsMo7, 2021-03-08:
> 

- [ ]

itsMo7, 2021-03-08:
Dd

dinesharjani, 2024-12-12:
Unfortunately we're quite limited regarding the Server functionality in the platform. We did entirely rewrite the "Server" portion last year (2023). Hopefully that improved the situation a bit.

closed 2024-12-12 by dinesharjani