nordicsemi/IOS-nRF-Connect #91: Buttonless DFU without bonds does not show up?

Laura-Zh, 2021-03-25:
Hello, I am trying to use the app to perform a firmware downgrade. nRF connect scans the device but I cannot find buttonless dfu without bonds anymore- it is not listed under attributes? I need this to complete the downgrade...how can I get it to show up?

philips77, 2021-03-25:
Hello,
It depends whether the new firmware that you flashed previously allows for
A) another update, meaning that it supports DFU
B) downgrading, by default it does not

Try restarting Bluetooth on your iPhone and connect again. Restarting Bluetooth will invalidate service cache on the phone, so on next connection a new service discovery will be done. If your device is bonded, try forgetting it and remove the bond information from the device, if necessary. Then connect again and try again.

Laura-Zh, 2021-03-26:
Ty. I don’t totally understand this stuff but was able to use an Android apk file to downgrade instead. The previous downgrade instructions for nRF connect did in fact work to downgrade the device but for whatever reason didn’t work again 🤷🏽‍♀️. Appreciate the reply.

closed 2021-03-26 by Laura-Zh