nordicsemi/IOS-nRF-Connect #94: nRF Connect for iOS  responds with 'Insufficient Authentication' error even if no service is added.

tnarik, 2021-05-31:
Hi,

I was testing a [bluezero](https://github.com/ukBaz/python-bluezero) based Raspberry Pi project, initially based on the `cpu_temperature.py` example from that repo.

It seemed to work well and I was able to read and write data from the nRF Connect **Android** version without pairing of any kind (didn't enable the pairing agent on the Pi).

I decided to give it a try with an iPhone and the latest iOS version of nRF Connect (installed 30/05/2021) and started getting the 'Bluetooth Pairing Request' pop-up when trying to connect to the same Pi. This was obviously failing because I didn't have any agent running on the Pi and pairing was not possible.

It seems at some point the nRF Connect iOS was responding with:

```
> ACL Data RX: Handle 64 flags 0x02 dlen 9                                                                                                                    #381 [hci0] 43.222765
      ATT: Error Response (0x01) len 4
        Read Request (0x0a)
        Handle: 0x0016
        Error: Insufficient Authentication (0x05)
```

, seemingly due to read requests from the Pi which do not happen in the case of Android. I'm not really sure what information to provide to try to debug this issue, if it is an issue on the iOS version, on the Android version, or somehow on the `bluezero` library and its dependencies
Enabling the default `bluetoothctl` pairing agent on the Pi does the trick, but I was wondering why is it needed for 'read' operations or no-operations at all, specially when i can still see the Temperature value if I tap on the read icon between modal popup refreshes.

Testing with `chrome://bluetooth-internals` on macOS works correctly as well (without the PI ever being paired to the computer).

philips77, 2021-06-02:
I tried this by setting similar GATT Server configuration in nRF Connect for Android and then connecting to it from iOS and had no such issue. iOS did not try to bond.
It may be caching issue, so perhaps try to turn off and on Bluetooth on iPhone and try again. Perhaps you used "encrypt_read" (or similar) flag before and iOS cached it?

tnarik, 2021-06-07:
This was happening even without characteristics advertised by the Raspberry, and due to iOS indicated that it required authentication 

It seems the issue is two pronged:
1) iOS advertises a BatteryService, which would require pairing if accessed
2) Bluez (from version 5.48), includes a `battery` plugin by default which would try to connect to the battery service.

So the peripheral device required either pre-pairing, implementing a security agent to handle that scenario as well, or disabling the plugin on the Pi, which is what I ended up doing and works nicely (see https://github.com/ukBaz/python-bluezero/issues/331 ).

Thanks for checking.

closed 2021-06-07 by tnarik

philips77, 2021-06-08:
Thank you for explaining the issue, we're glad it works.