nordicsemi/IOS-nRF-Connect #95: I keep getting 'GATT Service Listing missing, Please try again after connecting to the Internet...'

hatefk, 2021-06-18:
[image]

device: iPhone 8plus, OS version:14.4.2. Also I am connected to the internet, but wondering maybe this list should be downloaded when the app is installed. Or maybe there should be a button to go and manually download/update that list rather than having it to be automatically download.

dinesharjani, 2021-06-21:
Hi @hatefk, thanks for reporting this.
We fixed this bug late last week. We will try and get the update out this week to fix the issue. And yes, you can manually trigger an update from Settings, but a bug in the code prevents this from completing successfully, hence the issue.

hatefk, 2021-06-25:
Thanks @dinesharjani for looking into it, I really appreciate your efforts. I am wondering if the team had any chance to push to fix to production? last time I checked (couple of days ago) I was still having the same issues.

dinesharjani, 2021-06-28:
@hatefk  sorry, not released yet.

What I can do for you is that if you share an email address with me (you can also use the 'send feedback' option in nRF Connect to do so, I can add you to our beta test group in Testflight and release an early build there. You'd be able to get access to the fix now, instead of waiting for release.

closed 2024-03-05 by dinesharjani