nordicsemi/IOS-nRF-Connect #96: is the BLE library under nrfConnect available to others?

sdetweil, 2021-07-24:
I need to get the device name broadcast by my beacon. iosLocation Manager will not provide, but nrfConnect sees it

i saw the Android library, but really need ios

dinesharjani, 2021-07-26:
@sdetweil Hi,

Sorry to say but there is not at the moment. It has beeen brought up and thought about, but at the moment it's not available.

What we can do, and what we have done in the past, is try to help you out with your issue. Can you provide screenshots?

sdetweil, 2021-07-26:
thanks for the prompt reply. 

I need to be able to differentiate devices w the same uuid. 

one idea is to put the Mac address in the name field. useful on Android app, but not iOS.as iOS location manager does not provide the name field on didRangeBeacons

philips77, 2021-07-26:
Looks like you're using iBeacons. To differentiate iBeacons with the same UUID they need to have different Major or Minor numbers. The MAC address is not available on iOS, and there's no space in the Advertising Packet to add Local Name or any other structure. The application should only depend on Major and Minor numbers assigned to the beacon, and if you need to replace the beacon you simply create a new one advertising the same numbers. MAC address may change in each packet they send.

sdetweil, 2021-07-26:
I understand the 'intent', but that is not the solution design. I have multiple ibeacons with the same UUID AND Major/Minor in the same room.   they represent multiples of the same exact Object.   no beacon knows about the others. I can filter in rangebeacons on uuid/major/minor

imagine  you, i and some stranger have beacons on our keychain..  they have the same identifying  info, as they were bought from the same place.. how can I tell them apart..  I want to find YOUR keys, not go to each person and say, oh sorry, not you.. 

your code got my beacon name... so SOMEHOW it worked.. without connecting, as connecting causes my beacons to go into wifi mode. 
(beacons are using arduino's at the moment)

philips77, 2021-07-27:
If all your iBeacons are attached to keys, they should have the same Major number. But to distinguish your keys, from someone's else keys, they should have different Minor numbers. An equal Major and Minor numbers should only be used for undistinguishable things.
Core Location doesn't return beacon name, but the same device may advertise with more than just iBeacon packet. The rest is available using Core Bluetooth, however, it is not possible to match beacon returned from Core Location to a Bluetooth LE returned from Core Bluetooth, as iBeacon data is stripped from advertising data before returning to the app.

sdetweil, 2021-07-27:
fun times. 

all the scanners, yours included, somehow attached the name to the correct beacon. is that happy circumstance?

I need major/minor to filter my objects. but there are/could be multiple of the same major/minor in the same room.  with multiples of the major for sure. I don't have any more information available to separate them. 
but I need to
I was hoping name would be my saviour.

I suppose I could change minor to a complex value. high order byte the original minor, and low order byte a sequence number. 

ble filter on major
app code filter on minor

sdetweil, 2021-08-06:
changed my code, don't use major/minor in the ranging to filter.. filter in code..

closed 2024-12-12 by dinesharjani