nordicsemi/IOS-nRF-Connect #97: Reading a Characteristic after Writing a Characteristic does not show the correct value

ElreboCM, 2021-08-14:
### Test environment
Application Version 2.4.12
Test Peripheral: Arduino NANO 33
Test Sketch: https://github.com/arduino-libraries/ArduinoBLE/blob/master/examples/Peripheral/LED/LED.ino

### Problem
If the characteristic is written with a value using the app (e.g. the LED char from the sketch set to 1) reading again the characteristic shows still the old value. 

### Expected Behaviour:
After writing a value to the characteristic, reading it again should show the new value. 

### Side note:
I have tested in the same setting with the LightBlue App  (4.2.1) reading from the characteristic after writing the written value is received

philips77, 2021-08-15:
On iOS the value read and written to characteristic aren't using shared property. nRF Connect also displays them separately.

The value you read from a characteristic of descriptor depends on the firmware implementation. You may easily write one that always returns "Hello Kitty" whatever you write. It also may be time related, if you read the value to quickly, it returns the old one. There is a setting in settings to disable/enable automatic read after write. You may want to try that.

ElreboCM, 2021-08-15:
In this particular case it looks like the sketch was missing writing the value sent again to the characteristic.

  ```if (switchCharacteristic.written()) {
        if (switchCharacteristic.value()) {   // any value other than 0
          Serial.println("LED on");
          digitalWrite(ledPin, HIGH);         // will turn the LED on
          switchCharacteristic.writeValue(1);

        } else {                              // a 0 value
          Serial.println(F("LED off"));
          digitalWrite(ledPin, LOW);          // will turn the LED off
          switchCharacteristic.writeValue(0);

        }```

closed 2021-08-17 by philips77