nordicsemi/IOS-nRF-Connect #102: Can't upload firmware via DFU on iPhone

thinkOfaNumber, 2021-12-16:
iPhone 12 mini
iOS 15.1.1
nRF connect 2.4.13 build 104

Hi Nordic, this is a great app and a great chip! I can connect to the NRF52840 via nRF connect, but when I try and update the firmware via the DFU tab, it gets to the point of sending the firmware but fails before the progress bar shows any progress. nRF connect shows a little failure popup message, and the 52840 reboots into the bootloader (the USB MSC folder shows up). I've tried about 5 times (after reinstalling our firmware with the bledfu enabled) and the result is always the same.

[image]
(note some more messages appear in the log after this screenshot)

The same process on an iPad 8th gen (iOS 14.7.1), 2.4.12 build 39 works successfully on the same chip with the same firmware zip file. I've verified it works with some serial debugging output, but also the progress bar gets to 100% and nRF reports success.

I'm having trouble getting logs from the iPad as the "share" option keeps disappearing before I can use it, but not sure how much use it would be as there are no errors.

Here is the full log from the failed attempt on the iPhone:
```
File Name: firmware-2021-12-16.zip
Parts: 1
Size: 212 KB
Soft Device Size: Zero KB
Bootloader Size: Zero KB
[Callback] Central Manager did update state to: Powered ON
Connecting to mydevice...
centralManager.connect(peripheral, options: nil)
[Callback] Central Manager did connect peripheral
Connected to mydevice
Discovering services...
peripheral.discoverServices(nil)
Services discovered
Starting Legacy DFU...
Connected to mydevice
Services discovered
Legacy DFU Service found
Discovering characteristics in DFU Service...
peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
DFU characteristics discovered
Reading DFU Version number...
peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0100
Version number read: 0.1
Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
DFU Control Point notifications enabled
Application with buttonless update found
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
[Callback] Central Manager did disconnect peripheral
Disconnected by the remote device
Connecting to mydevice...
centralManager.connect(peripheral, options: nil)
[Callback] Central Manager did connect peripheral
Connected to mydevice
Discovering services...
peripheral.discoverServices([00001530-1212-EFDE-1523-785FEABCD123])
Services discovered
Legacy DFU Service found
Discovering characteristics in DFU Service...
peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
DFU characteristics discovered
Reading DFU Version number...
peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0800
Version number read: 0.8
Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
DFU Control Point notifications enabled
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Writing image sizes (0b, 0b, 211888b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x0000000000000000b03b0300, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
Data written to 00001531-1212-EFDE-1523-785FEABCD123
Start DFU (Op Code = 1, Upload Mode = 4) request sent
Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100101
Response (Op Code = 1, Status = 1) received
Writing Initialize DFU Parameters...
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x0200, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Writing to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x5200ffffffffffff0100b600dcfc, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x0201, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Data written to 00001531-1212-EFDE-1523-785FEABCD123
Data written to 00001531-1212-EFDE-1523-785FEABCD123
Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100201
Response (Op Code = 2, Status = 1) received
Initialize DFU Parameters completed
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x080c00, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Data written to 00001531-1212-EFDE-1523-785FEABCD123
Packet Receipt Notif Req (Op Code = 8, Value = 12) request sent
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x03, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Data written to 00001531-1212-EFDE-1523-785FEABCD123
Uploading firmware...
Sending firmware to DFU Packet characteristic...
Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100306
Response (Op Code = 3, Status = 6) received
Error 6: Operation failed
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x06, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
[Callback] Central Manager did disconnect peripheral
Disconnected by the remote device
```

Thanks!

dinesharjani, 2022-01-11:
Hi @thinkOfaNumber , sorry for the delayed response.

Would it be possible for you to send me the DFU image for me to test this for myself and check the DFU library logs and find out the true issue?

thinkOfaNumber, 2022-02-02:
Now it's my turn to apologise for the delay! I am no longer working on this project and no longer have access to the hardware... @robsonos might be interested in following this up though.

closed 2022-05-18 by dinesharjani