nordicsemi/IOS-nRF-Connect #103: Settings Section Stuck in Loop

AppendUser, 2021-12-30:
Application version 2.4.13
Build number 104
Device iPhone 11 running iOS 15.2

Sometimes when I go to the settings menu, the app will go back and forth between the overall settings page and the General > Scanner settings page. When this happens I am unable to click on anything to make it stop other than accessing a separate section/menu using the toolbar on the bottom. Force closing the app will eventually cause it to stop so I can use the settings section normally, but not always on the first try.

dinesharjani, 2022-01-25:
Hi @AppendUser,

Sorry for the delay. I looked into this a few days ago, but could not reproduce the issue. Do you have any steps to make this happen?

dinesharjani, 2022-01-28:
This is fixed and will be part of the upcoming 2.5 Release. Feel free to get in touch to try out a beta with the fix on TestFlight.

closed 2022-01-28 by dinesharjani